yunnanpl/esp32_python_eq3 #7: Missing setup instructions

tofferlicious, 2021-10-05:
Thank you very much for this project. Unfortunately I could not get it running properly so far.

I configured everything to my best knowledge inside the secret_cfg.py.
But somehow I am still getting this exception and I cannot further analyze it, due to the "simple2.mpy" binary:

`Traceback (most recent call last):
  File "main.py", line 888, in <module>
  File "main.py", line 860, in fmqtt_recover
  File "simple2.py", line 70, in connect
MQTTException: 25`

Am I still missing something? I got Home Assistant running with the add-on Mosquitto broker and I am using the credentials from it.
If I remember correctly, this was working fine with the esp32_mqtt_eq3 project.

yunnanpl, 2021-10-10:
Dear tofferlicious,
I have not noticed it either, as I do not use password with mqtt.

Unfortunately, as it was "corona"-project, I do not have so much time as I had before.
I do use it with home-assistant, so it surely works. The configuration is the same as it was for esp32_mqtt_eq3,
Still, I have not finished the "auto mqtt home assistant" configuration for the climate components.

Yes, debuginng is quite nice, explicitly with webrepl console (if you have issues with it, try to open the console in incognito mode).
There is still a lot work to do ;) but the stability is much better (the esp32 are rebooting themselves if webpage, mqtt or anything fails).

PS. I find the story with ble messages triggering an interrupt slightly confusing... so even if I tried to comment the code for myself, it might be challenging to read it :D

yunnanpl, 2021-11-22:
So, as discusses, the code is improved significantly. Much faster, and even more stable.
I also added the home assistant autodiscovery, which makes it really fun to use (no climate and mqtt manual configuration...).
Now all automations are easy and simple, and everything is working even on weak BLE connections !

Still no "modes" added, but the code is ready for it, so it will be added soon (at least the auto, boost and manual).
The on, off, away modes can be easily "simulated" with temperature setting.
Furthermore, for me the offset is important, so I will add it both on the EQ3 part, and an additional offset on ESP32 (as my feeling is that the EQ3 offset does not work so well).

yunnanpl, 2021-11-24:
So, the newest version is really quite stable and robust, even for EQ3 thermostats with RSSI worse than -90.
Autodiscovery works very well.
Instructions and change log added to separate files in GitHub for readibility.

closed 2021-11-24 by yunnanpl