yunnanpl/esp32_python_eq3 #14: Version of CalorBT software

lechup, 2022-02-12:
What version of sw are You running on your valves? I'm still on 1.1 or even older, just thinking, should I upgrade or your fw should work with any sw version?

yunnanpl, 2022-02-13:
I am surely on an older version (I will check on which one). I would rather recommend not to upgrade, at the moment.

I know that some newer fw versions require BLE authorization, which was an issue in previous micropython version (was not available). Now the BLE authorization should be in micropython, but I have not worked on it yet.

I have 9 thermostats, so if I upgrade one, I have to upgrade all. So I clearly need to book a reasonable time slot for this ;)

lechup, 2022-02-14:
Ok, if You will find some time just check on one of valve what they display while turning on. And You can put info in main readme that is is tested on this version of firmware. I'll let You know what I have.

yunnanpl, 2022-03-19:
So, it seems that on Android it is not possible to find out what version do I have.
**Still, I do NOT have the version where the 6-digit key is needed for pairing,**
(the new version is surely more secure, as wished by the community since longer time, I just have not had time to have a look at it).

yunnanpl, 2023-01-23:
I was exchanging batteries, and I finally checked what the **Firmware of my EQ3 is, and it is 1.20** (so the one without 6 digits pairing).

wieluk, 2023-03-01:
Hey i just bought another eq3 but this one acts weird. It reports right Temperaturen but i can t control it. Do you know what could be the problem? All 4 other give me a 6 digit pairing key but the new one only has a 4 digit key. How can I check the Version?

yunnanpl, 2023-03-01:
All of mine have the 4 digit key, as I still use (happily) the old firmware (1.20). So yours in theory it should work fine with my "software".
I think, when you put the battery, then **one of the values appearing on the LCD screen is the firmware version during initialization**. This is how I obtained my firmware version.

PS. What do you mean, that it acts weird ?

yunnanpl, 2023-03-02:
Hi @wieluk: that is a good news.
I have not updated my thermostats, as I do not wanted to break everything. My 9 thermostats work with 1.2 firmware. Still, if it does work with newer too, maybe it is time to upgrade... :)