yunnanpl/esp32_python_eq3 #15: boot.py throws error: ValueError: incompatible .mpy file

Eule01x, 2022-09-28:
Hi,
this is the first time, that I'm using a Micropython project on my ESP.
But finally I was able to flash the ESP32, upload the 4 files and connect via putty...
When I then try to start, I get the following messages:

`Traceback (most recent call last):
  File "boot.py", line 17, in <module>
ValueError: incompatible .mpy file
Traceback (most recent call last):
  File "main.py", line 35, in <module>
NameError: name 'config2' isn't defined
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32`

Seems like he's unhappy with the simple2.mpy file?

Eule01x, 2022-09-28:
I found out, the mpy version is incompatible with the actual firwmare version [v1.19.1](https://micropython.org/resources/firmware/esp32-20220618-v1.19.1.bin)
So I rolled back to the previous one: v1.18

yunnanpl, 2022-09-29:
Hi Eule01x,
winter is coming... I expect more questions ;) First, thanks for using it, and thanks for feedback !
I am using mostly python everywhere... as I am just bad at anything else.

Concerning your query, I noticed new micropython, but had no time to test it. The MPY files are pre-compiled, to speed up booting and save memory (mostly RAM). Still, this is only my hunch, as I have no idea if this changes anything at all.

In theory, you can just throw the simple2.mpy file away, and download the original one, and this should work.

- https://github.com/fizista/micropython-umqtt.simple2/blob/master/src_minimized/umqtt/simple2.py

There is slight danger, that it is not the pre-compilation that makes the issues, but the simple2.py library itself (maybe it uses some functions which became obsolete in micropython 1.19...).

yunnanpl, 2022-11-07:
So, it seems that just deleting the simple2.mpy file and uploading simple2.py (not pre-compiled) works already on the newest micropython 1.19.
I also uploaded the newer version of the script, with some updates.

closed 2022-11-08 by yunnanpl