yunnanpl/esp32_python_eq3 #17: minimum temp, valve status, boost

wieluk, 2022-10-20:
Hey, I have some questions for your great work. Is it possible to set boost mode via homeassistant or to turn the trv's on/off? Right now the minimum I can set them to is 12°C. On the Original eq3bt integration for homeassistant I had a value how far open the valve is (0-100) is this also possible here? 

Is this just because of mqtt autodiscovery or is it not possible with your work?
With autodiscovery the climate entity looks like this:
```
hvac_modes:
  - heat
min_temp: 12
max_temp: 28
target_temp_step: 0.5
current_temperature: null
temperature: 12
friendly_name: Bath
supported_features: 1
``` 

Thanks in advance for any help.

wieluk, 2022-10-20:
I found something for the min and max temperature in line 1258 of main.py

```        
  devicec = '{ "name": "clim_' + mac + '_eq3", \
      "unique_id": "id_' + mac + '", \
      "modes": [ "heat" ], \
      "mode_cmd_t": "' + config2['mqtt_eq3'] + mac + '/radin/mode2", \
      "mode_stat_t": "' + config2['mqtt_eq3'] + mac + '/radout/status", \
      "mode_stat_tpl": "{{value_json.mode2}}", \
      "temp_cmd_t": "' + config2['mqtt_eq3'] + mac + '/radin/temp", \
      "temp_stat_t": "' + config2['mqtt_eq3'] + mac + '/radout/status", \
      "temp_stat_tpl": "{{value_json.temp}}", \
      ' + currtemp + ' \
      "min_temp": "12", \
      "max_temp": "28", \
      "temp_step": "0.5", \
      "hold_state_template": "{{value_json.mode}}", \
      "hold_state_topic": "' + config2['mqtt_eq3'] + mac + '/radout/status", \
      "hold_command_topic": "' + config2['mqtt_eq3'] + mac + '/radin/mode", \
      "hold_modes": [ "auto", "manual", "boost", "away" ] }'
```
The min temp should probably be 5 and max 29.5.
But I don't really understand the "hold_modes" can you tell me how this works?

yunnanpl, 2022-10-22:
Hi wieluk,
unfortunately it is not possible to switch on/off/boost/away. I have never used it with previous implementations, so I just skipped it (at the moment). Furthermore, there were different modes names in climate implementation than the ones expected by the eq3, so it was easier to skip those, than to rewrite "climate".
The hold_modes are really something what is integral for "climate" implementation, so I added those, to avoid warnings, but it does not do anything.

Concerning the 12-28 C range, I do not remember why I choose it. It may be that it was not working well below this value (which may have been my mistake). I treat 12 C as closed (OFF), and 28 C as BOOST, and it works well in home assistant.
You can easily change the values to the 5 and 29.5, and I can change those in new version (assuming it will work well).

First I thought that you can set the valve open value from 0-100... setting is not possible (but would be nice).
Reading it is possible, and I recall there was such value in the Bluetooth packet. Still... I was not sure what it will be good for ;)
I see no problem adding it, as I also plan to add the battery status (which I personally find more important).

wieluk, 2022-10-22:
Thanks. If anyone wants to know you can just change the min and max temperaure in line 1258 of main.py
but you also need to change line 168 to the new temperature like this.
```
elif addr[0:8] == '00:1A:22' and data1 == 'settemp' and (str(data2).split('.')[0]).isdigit():
    data1 = 0x0411
    if float(data2) > 29.5:
        data2 = 29.5
    if float(data2) < 5:
        data2 = 5
```

For me the state of the valve was very important because it shows you how much each radiator heated. You can just use a integration sensor in homeassistant or use the utility_meter to compare them over time.

wieluk, 2022-11-01:
Is it possible for me to do it on my own? Could you show me a direction where to start because I am kind of lost in the code

yunnanpl, 2022-11-01:
I will look at it in a near future.
Getting additional data (like valve open state, or battery status) is not so simple, as you first need to decode the correct bytes from the bluetooth packet (there are other eq3 software's which describe it in details, and which I refer to on the main page).
Then you need to put it in the mqtt package, and clearly correct the home assistant autodiscovery mqtt packet.

So, unfortunately there will be multiple changes in the code, and I do not remember from the top of my head.

wieluk, 2022-11-06:
Okay, thanks. I think it would be awesome to have all these features. In [this](https://github.com/softypit/esp32_mqtt_eq3#supported-commands) project there are a lot of supported commands. Even like offset.

yunnanpl, 2022-11-06:
I can guarantee that NOT all features will be implemented, as many are slightly useless :)

So, ON, OFF, BOOST, AWAY... AUTO or MANUAL can be all controlled from HomeAssistant.
Furthermore the HomeAssistant Climate "object" does not foresee many of there settings, and I will not start to change HomeAssistant now.

I did changed the temperature range to be 5 to 29.5. I rechecked the script to work with Micropython 1.19 (it does).
I do read the valve open value (hopefully correctly ;)). I do send the value with MQTT, but the autodiscovery in HomeAssistant has to be set manually (at the moment) (and HomeAssistant climate object does not expect it, so it will be a separate entity anyway).

I try to read the battery value, but it is really only GOOD or LOW, so I need to wait to see which byte changes.

Still, I tried to optimize the main loop, which broke the system, but I will have look at it.
So, now I am fixing the things I broke ;D

wieluk, 2022-11-06:
Thanks for doing this. I just read a little about the mqtt autodiscovery and I think you can add extra attributes with the  [json_attributes_topic](https://www.home-assistant.io/integrations/climate.mqtt/#-----configuration-variables--). Here is a config [example](https://www.home-assistant.io/integrations/sensor.mqtt/#json-attributes-topic-configuration). Just if you want to do it :)

yunnanpl, 2022-11-07:
I am not too good at it but I think it does not work this way. I clearly use this to read the mqtt values, but this does not add special attributes to the object.
The climate object, does not have any attributes for battery or valve open value, so a separate sensor needs to be created.
Maybe it is possible to connect this sensor to the climate object, but I need read and test it.

At the moment, the esp32 just sends the valve open value in an mqtt packet, which can be used.

wieluk, 2022-11-08:
Oh that is weird because the built in eq3bt integration had the valve value as attribute but thanks for the release.

wieluk, 2022-11-08:
Sorry I meant the homeassistant integration that used to work before they updated python to 3.10: 
[https://www.home-assistant.io/integrations/eq3btsmart/](https://www.home-assistant.io/integrations/eq3btsmart/)

wieluk, 2022-11-08:
I have a question about the valve value. For me the value is always 4
for this example where it should be 0
```
{
    "trv": "MAC",
    "temp": "18.0",
    "valve": "4",
    "battery": "NOT_IMPLEMENTED",
    "mode": "manual",
    "mode2": "heat"
}
```
and for this one where it probably should be 100 or atleast 80 because I hear how it opend and the water flows.
```
{
    "trv": "00:1A:22:0E:AC:EE",
    "temp": "25.0",
    "valve": "4",
    "battery": "NOT_IMPLEMENTED",
    "mode": "manual",
    "mode2": "heat"
}
```

yunnanpl, 2022-11-08:
Correct, I took the wrong byte. Now fixed.
Also battery and valve create separate sensors in homeassistant automatically.

wieluk, 2022-11-09:
Thanks for your work. What value would I expect in homeassistant for a low battery? not OK?

yunnanpl, 2022-11-09:
I do not use the "battery" sensor type, as this expect percent values.
I think the battery binary sensor is ON (for low battery, alarm ON) or OFF (for good battery), but I found it not intuitive.

Now the values for battery are OK or LOW (it is a none type sensor).

wieluk, 2022-11-09:
My esp32 crashed sometimes lately but it seems to be fixed for now. I added "exp_aft": 3600 to the valve sensor so I can get a notification if homeassistant doesn't get a mqtt value in one hour. Just wanted to let you know if you think it is something with value for the project.

line 1359:
```
            devicec = '{ "name": "clim_' + mac + '_eq3_valve", \
"state_topic": "' + CONFIG2['mqtt_eq3'] + mac + '/radout/status", "exp_aft": 3600 ,"value_template": "{{ value_json.valve }}",  \
```

yunnanpl, 2022-11-09:
Yup... crashing is an issue, as stability is the main aim of this project.
Changing things quickly, usually breaks some other stuff.
Unfortunately I will have no time to focus on it in some weeks.

wieluk, 2022-11-10:
yea unfortunately I downgraded to the old version because the new one crashes a lot sometimes it recovers and sometimes it does not.

yunnanpl, 2022-11-10:
Yes, but a good news is, that I checked it yesterday, and my "watchdogs" complained about low memory (and I had some "improvements" which were really not one...).
Now it is running stable (12 hours now) on my 4 ESP32, on micropython 1.19 and 1.17 (as some of my devices are nicely hidden, so I can hardly access those 😄).

I will upload it in the evening (when the "tests are positive").

wieluk, 2022-11-10:
I think you don't really need it on all sensors I just wanted to notify my phone when the esp32 is down.

yunnanpl, 2022-11-10:
Wait wait ;)
It should NEVER happen, that the esp32 will die and become unresponsive. The whole idea of this project is, that I was unhappy with other projects, so I created this one.
The main aim is that it will be rock stable... and if not, then at least it will automatically recover after crash or reboot.

Until now, the stability was weeks to months for some devices. and (to my knowledge) no temperature change request was ever lost.

I was also doing some fixes outside of esp32, to fix its instability or unresponsiveness... so if you feel you need to do it right now, then this project have gone completely bad.

In short:
- if wifi is lost, or mqtt is lost for a longer time (5 minutes or something) then "recovery" function is triggered (and finally a reboot is requested)
- if the webpage is not responsive, then reboot is requested
- if device is not responsive for some time (4 minutes) then watchdog is doing hard reboot (due to flood, dos, or something)

After the reboot, mqtt messages should be pulled, and everything should recover and run smoothly again.

I know the current script is slightly low on memory, but a code cleanup (objects, etc.) can take longer time, and not sure if it will lower the memory usage.
Still, additional functions will surely only worsen the memory stats.

PS. I will upload a new version in the moment. It was running rock stable on two test devices with micropython 1.19.

wieluk, 2022-11-12:
Thanks the new version works for me!
How did you add other bluetooth devices like temperature sensor? I am thinking about trying to add the Qingping CGH1 window/door sensor. Right now I have them added with [passive ble monitor](https://custom-components.github.io/ble_monitor/by_property#status) integration in homeassistant but the commection range is not that good. Maybe when I have more time and understand more of the code i can try to.

yunnanpl, 2022-11-12:
I am happy that it works for you.
My 4 ESP32 and 9 EQ3, are working very stable since 3 days, so I hope the issues are solved.

Yes, I do support my Xiaomi thermometers. It is quite bad, as those are not even passive (I need to send request).
I do have smaller thermometers now, which are passively sending info. I have not implemented those yet.
Still, it is not hard, if the BLE packet is understood (described).

I opened an "issue" for this, to close this discussion:
https://github.com/yunnanpl/esp32_python_eq3/issues/20

closed 2022-11-12 by yunnanpl