yunnanpl/esp32_python_eq3 #18: ntptime OSError: -202

wieluk, 2022-11-07:
Hey, today my esp32 was not responding anymore so I checked what was going on and at boot I get this error:
```
Traceback (most recent call last):
  File "boot.py", line 132, in <module>
  File "ntptime.py", line 20, in time
OSError: -202
- load wl failed, setting default, this is ok  [Errno 2] ENOENT
= f mqtt recover
Traceback (most recent call last):
  File "main.py", line 1310, in <module>
  File "main.py", line 1291, in fmqtt_recover
  File "simple2.py", line 223, in connect
OSError: [Errno 113] ECONNABORTED
```
I already tried a lot of different ntp settings even IP-server and reflashing the esp32. Do you know what this could be?

wieluk, 2022-11-07:
seemd to be a dns error because of adguard sorry.

closed 2022-11-07 by wieluk

yunnanpl, 2022-11-07:
No problem. It looks like network issue (as there is mqtt connection issue too), so adguard would be a reasonable explanation.