yunnanpl/esp32_python_eq3 #23: Should this work without internet acces?

wieluk, 2023-03-02:
Hey sorry for the all the question but I just moved and I don't have internet right now but local wifi.
I noticed that your project relies on a ntp-time server. I tried to use my qnap nas as local time server but can't get it working.
Do you know what could be going wrong? I just used this is the secret_cfg.py:
```CONFIG['ntp_host'] = '192.168.0.37'```

but I get this error
```= ip 192.168.0.32
= time -3155673600
= mqtt ping 17
+ booted
= f mqtt recover
= send mqtt autodiscovery
= set time ntp
Traceback (most recent call last):
  File "main.py", line 1, in <module>
  File "main.py", line 1, in fntp
  File "ntptime.py", line 38, in settime
OverflowError: overflow converting long int to machine word
```

yunnanpl, 2023-03-02:
Hi @wieluk, it should clearly work without internet. I use it without internet, as all my IOT devices have no direct access to the internet.
I have NTP server on my Fritz and on my Odroid Linux. Whereas the ESP32 take the time from Fritz.
**I must say that it is kind of unusual that the time is negative in your log...**

The line "= time" shows the time obtained from NTP (**print('= time', ntptime.time() )**).
The message "= set time ntp" comes from the function in main.py:
```
def fntp() -> None:
   print('= set time ntp')
   VGLOB['timentp'] = time.time()
   ntptime.settime()
   return
```
So, it seems there is an issue with the readout of time.time(), or the ntptime.settime() function.

The time.time() and ntptime.time() on my system gives (as expected) the same value:
```
>>> time.time()
731095529
>>> ntptime.time()
731095535
```

So, if possible, start the functions:
```
ntptime.settime()
```
This should run without any warning, nor response.
Then check if these both commands give the same value:
```
time.time()
ntptime.time()
```

Finally, check if the NTP on QNAP is working fine, and if the time is set properly. If QNAP does not have access to the internet, it may have wrong time too...

yunnanpl, 2023-09-29:
So, with the newest version, if the local (or set in settings NTP) fails, then there is a fallback to european NTP server.

https://github.com/yunnanpl/esp32_python_eq3/commit/05ed9d60cb3d02cc791c9e1065d3939baa6fd9c0

closed 2023-09-29 by yunnanpl

wieluk, 2023-10-04:
But if it uses a european NTP server it will still crash when I block outgoing connections in my router right? 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
= ip 192.168.0.33
Traceback (most recent call last):
  File "boot.py", line 147, in <module>
  File "ntptime.py", line 25, in time
OSError: [Errno 116] ETIMEDOUT
MemoryError: memory allocation failed, allocating 1113 bytes
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
```

For me this happens when I block outgoing connections.

yunnanpl, 2023-10-05:
Yep, I understand now.
Clearly I assume that local NTP is working. Secondly, as I write it myself, I kind of trust it ;) so I allow one direction access to the internet NTPs just in case. Now I removed internet NTP connection (only if asked for).
So, now, to mitigate the issue, the system will save a timestamp in a file every 12 hours (should not be an issue for the memory), which can be recovered when the system reboots, so there will be not more than 12 hours time error.