yunnanpl/esp32_python_eq3 #24: Discovered MQTT entities with a name that is equal to the device name

wieluk, 2023-08-03:
Hey I just wanted to let you know that this warning came with latest homeassistant update.

[image]

But it will work until 02.2024.

yunnanpl, 2023-09-07:
Yes... I have issues with so many different names.
Furthermore, I would assume that device name is less important, while the entity name is critical... and they set it to "null" as workaround ;)
Still, I will make it before Christmas, no worries :)

Thanks for the memo ! I do not upgrade HA too often, as it breaks my home-made customization... so I need to reserve an reasonable amount of time for this.

yunnanpl, 2023-09-29:
After some trying, I changed the device names (and IDs) by adding _dev and _id respectively. Hopefully it will work without errors now.
Still, I have not touched the main "name" and main "id" so I required no changes in my scripts or GUI. Hopefully it will be the same for you.
See https://github.com/yunnanpl/esp32_python_eq3/commit/05ed9d60cb3d02cc791c9e1065d3939baa6fd9c0

closed 2023-09-29 by yunnanpl