yunnanpl/esp32_python_eq3 #25: Exception raised when updating state of sensor.clim_bedroom_battery

wieluk, 2023-09-16:
Hey there the battery status of the trv is not working anymore because homeassistant seems to except a int for the sensor.

```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 586, in state
    numerical_value = int(value)
                      ^^^^^^^^^^
ValueError: invalid literal for int() with base 10: 'OK'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 589, in state
    numerical_value = float(value)
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'OK'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/models.py", line 305, in process_write_state_requests
    entity.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 730, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 830, in _async_write_ha_state
    state, attr = self._async_generate_attributes()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 771, in _async_generate_attributes
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 736, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 593, in state
    raise ValueError(
ValueError: Sensor sensor.clim_bedroom_battery has device class 'battery', state class 'None' unit 'None' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'OK' (<class 'str'>)
```

Do you know if i have to do something in homeassistant itself or in your code?

yunnanpl, 2023-09-18:
Hi wieluk,
that is weird, I will have a look at this on weekend. I think the EQ3 sends a binary value (good/bad), but at the time the values OK were completely accepted (and recommended) by HA. Now, the only thing to do is to set "100" (%) as OK, and... 0 (???) as bad battery. Clearly, there are no values in between.

yunnanpl, 2023-09-29:
Yep, now new or acceptable battery will show value 100, and low battery will show 0 (to maybe trigger some HASS alert or something). Still, if EQ3 responds low battery it will still live for some weeks.
See https://github.com/yunnanpl/esp32_python_eq3/commit/05ed9d60cb3d02cc791c9e1065d3939baa6fd9c0

yunnanpl, 2023-09-29:
Closing the issue.
Still, I am sending a number as a string. So if HASS will complaint, just reopen it, and I will fix it again.
(I have not updated my HASS yet, so I cannot check it myself).

closed 2023-09-29 by yunnanpl