yunnanpl/esp32_python_eq3 #27: allow setting temp to 4.5 to save battery

wieluk, 2023-09-30:
hey thanks for the update! On my version I did allow setting temp to 4.5°C instead of 5°C it doesn't seem like a big change but I think it could save a lot of battery in the summer because it turns the eq3 off. When the thermostat is set to 5°C it still calibrates regularly. I am not 100% sure but i think if you set it to 4.5 the display shows "off" and until now none of my 5 did a calibration run while off.

It is just a simple change in the main.py in line 233 and 234
```
        if float(data2) < 4.5:
            data2 = 4.5
```

and in line 1492
```
' + currtemp + ' "min_temp": "4.5", "max_temp": "29.5", "temp_step": "0.5", \
```

yunnanpl, 2023-10-01:
Hi, when it is so, that would be nice.
I have not implemented the other EQ3 functions due to lack of time, and furthermore those do not correlate well with HASS HVAC.
I do not see issues with battery, but the recalibration noise is sometimes annoying.

So, it is reasonable to check if 4.5 C setting switches the EQ3 off, and use it.

yunnanpl, 2023-10-03:
Yes, it works (seem so) !
Setting temp to 4.5C sets the EQ3 to OFF, and setting temp to 30.0C sets the EQ3 to ON (full power, kind of BOOST).
Now both settings are allowed, and working.

I did some other improvements, and maybe I will slow the responsiveness by 0.5-1 sec, to improve stability (and lower battery consumption).

I will upload new version soon, after some basic tests.

yunnanpl, 2023-10-03:
Now, just a thought.
Assuming it is summer, we set the EQ3 to 4.5 (-> OFF) to avoid re-calibration, and therefore battery loss...
Maybe for the thermostats which are OFF, the delay between automated queries should be longer too (twice ?) ?
BT connections drain battery too.

In the lines 1422, the automated adding of "manual" command is done. Using this command the current status of thermostat is obtained.
One possible solution would be to save the previous valve opening value, and if it was zero in previous run, then the run time will be doubled. This would save the battery always, when low flow is detected...

If valve flow is 0, then auto-checking time is doubled (instead of 60sec, 120sec).
Additionally, if thermostat is set to OFF, then the time is doubled again.
Just need to implement it cleanly ;)

wieluk, 2023-10-04:
Yea it is a good idea. But I think I would not make the query longer if valve is 0. Because often the valve is not open that long and I would like to know when the valve opens.

yunnanpl, 2023-10-05:
I have one annoying thermostat, with low battery and low signal level, and it was making my ESP32 crazy.
Now I implemented some changes (a lot of changes), to avoid crazy thermostat from rebooting the ESP32.
I will do some tests, and implement the delay in querying.

yunnanpl, 2023-10-16:
So, it seems that the code size is at the limit, so adding extra stuff often ends with "memoryerror: memory allocation failed".
As the potential for optimization is huge (as it is a slight mess), I am cleaning the code right now (testing right now).
I am thinking about exporting some bits and pieces to separate libraries, which can be loaded and unloaded if necessary.

I will post something after a few days of testing.

yunnanpl, 2023-10-20:
So... a lot has to change, that everything stays the same ;)
Due to seemingly too long code, I started to do code optimization. Doing it I found some weaknesses and not caught exceptions, which now work much better. Recovery from mqtt server issues, or bad/broken ble connections are also better now.
Even though it rarely happened, it should never happen that the script will stuck without answering (reboot is better than a stuck device).

In short, the code is better, faster and cleaner, but this feature is still not implemented.
I think, as said above, I need to split the code into pieces, in order to be able to add new things, otherwise the memory is overloaded.

Still, new release is available, without much fuss :)

yunnanpl, 2023-11-13:
Finally, some significant code cleanup is done. I did managed to export part of the code to the other file, which somehow improved the available memory.
Now, if the TRV is set to 4.5 (means OFF), then it doubles the query time. Now testing stability.

In addition, I shortened the watchdog time, to avoid situation where the ESP becomes unresponsive (it happened for one thermostat which behaved strange due to low signal). Now it is better to reset in worst case. I also tried to make it more WWW flood immune, but it is tricky.

Now, I think about not sending the MQTT message if no change in data happened. Hopefully this will lower the load on the Home Assistant database.

I will push the commit after a few days of testing.

wieluk, 2023-12-05:
did you push the code cleanup yet? my esp is dying from time to time maybe this would help. There is also one trv with low signal.

yunnanpl, 2023-12-05:
Soooooo.... a lot of testing ;) Check this out.
https://github.com/yunnanpl/esp32_python_eq3/tree/2679b3ee142045e23f78dfd1b9585783b1aa3660

There are a lot of changes, but the most critical ones:
- export of some functions into separate file main2.py to save memory and keep the code cleaner
- significant changes in webpage serving, as sometimes flooding led to device reset (this does not happen now)
- added 2 times delay for TRVs which are OFF
- improved logging which is now clearly visible in info tab, whereas shell is not spammed (overwrite of print function)
- removing some bits and pieces from boot.py if not really necessary
- change in connection behavior, now the connection is initiated once, and then system waits (before it was trying to connect multiple times in one round)

Update:
One extra file is added main2.py, which consists of functions exported from main.py.
boot.py and main.py should be updated too. Other files are the same as before.

So, devices with RSSI of around 85 should almost always work without issues. Above this values it can be tricky.

Give a try and let me know if it got better.

Surely, in case of unexpected issues like bad RSSI, lack of WIFI, problems with MQTT... the device cannot recover well, therefore there is a reboot function. I want to add recovery from reboot (so, like work list saving), but this is in progress now.