yunnanpl/esp32_python_eq3 #29: Hiii, I am trying for LONG RANGE BLE (>300meters) using ESP32 C3 mini-1.

Saiuday9787, 2023-11-13:
Hiii, I am trying for LONG RANGE BLE (>300meters) using ESP32 C3 mini-1.

I tried esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P21);  at server side and client side, still i am able to get Line of site upto 100meters and able to tx via one floor concrete slab. still my requirement is not satisfied, 
Can you suggest on how to achieve above 300meters range.

_Originally posted by @Saiuday9787 in https://github.com/yunnanpl/esp32_python_eq3/issues/22#issuecomment-1807495929_

yunnanpl, 2023-11-13:
So, I do not think that this is a correct place for such discussion ;) as this does not concern this project much.
Furthermore, keep in mind that BLE is not really made fur such things, as it would be not easy to get WIFI at such distance.

Nevertheless, there are some Long Range BLE project in the internet. So in this case, your board may be a bad choice as it even does not have an dedicated antenna port. I cannot tell much about gains, as I do not use it, but I do own this one (ESP32 Cam with external antenna port): https://randomnerdtutorials.com/esp32-cam-connect-external-antenna/
There is IPEX to SMA (or something more standard) adapter needed, and then some directional antenna can be attached easily.
(IMPORTANT: both the "sender" and "receiver" have to be equipped with such directional antennas, to increase stability and range, one side is probably not enough).

I think for long range applications some other approach would be much better. For example LoRa, which can be also obtained as ESP32 with buiilt in LoRa transmitter, or added separately:
https://randomnerdtutorials.com/esp32-lora-rfm95-transceiver-arduino-ide/
The range is much higher due to lower frequencies, and power consumption is also lower, clearly both at a cost of transmission speed.

Saiuday9787, 2023-11-17:
Understood But My requirement is with ESP32 C3 only, I just wanted to know what is maximum range I can achieve. Can any one suggest me steps/values to configure Devices in order to Get max range.

Currently I have configured my GATT Client as below:
static esp_ble_ext_scan_params_t ext_scan_params = {
    .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
    .filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_duplicate = BLE_SCAN_DUPLICATE_DISABLE,
    .cfg_mask =  ESP_BLE_GAP_EXT_SCAN_CFG_CODE_MASK,
    .uncoded_cfg = {BLE_SCAN_TYPE_PASSIVE, 0x40, 0x40},
    .coded_cfg = {BLE_SCAN_TYPE_PASSIVE, 0x40, 0x40},
};
const esp_ble_gap_conn_params_t phy_coded_conn_params = {
    .scan_interval = 0x40,
    .scan_window = 0x40,
    .interval_min = 320, 
    .interval_max = 320,
    .latency = 0,
    .supervision_timeout = 600,
    .min_ce_len  = 0,
    .max_ce_len = 0,
};

GATT server configured as below:
esp_ble_gap_ext_adv_params_t ext_adv_params_2M = {
    .type = (ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE|ESP_BLE_GAP_SET_EXT_ADV_PROP_INCLUDE_TX_PWR),
    .interval_min = 800,	//1600,//0x20,	//value*0.625millisec
    .interval_max = 800,	//1600*4,//0x20,
    .channel_map = ADV_CHNL_ALL,
    .filter_policy = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy = ESP_BLE_GAP_PHY_CODED,	//ESP_BLE_GAP_PHY_1M
    .max_skip = 0,
    .secondary_phy = ESP_BLE_GAP_PHY_CODED,	//ESP_BLE_GAP_PHY_2M
    .sid = 0,
    .scan_req_notif = true,//false,
    .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
    .tx_power = 126,//EXT_ADV_TX_PWR_NO_PREFERENCE,
};

yunnanpl, 2023-11-19:
Hi @Saiuday9787, usual range of such kind of BL/BLE devices is 1 up to 10 meters. The class 3 devices can in theory go up to 100 meters, but this means that both of the BL/BLE devices has to be this class (or better ;)). Nevertheless, my experience is that the range is around 10 meters anyway (stable), and the reason is that the issue is not necessarily the ESP32, but the device you are talking to !
The issue is, that fixing the range or antenna on the closed receiver device (like EQ3 TRV) may be very tricky.

So in short, you probably cannot do much with software only, but it has to be a hardware change (larger antenna, maybe directional, on both ends of the connection). So do more with hardware, as you will get much more gain from this, than from fiddling with the software.

My experience is that the BLE is fine around 10-20 meters (slightly less through walls) and with quite weak "receiver" device. I think it will be really good if you get more than 20-30m stable connection without some hardware upgrades.

Anyway, there are much better sources for range improvement ;)

Saiuday9787, 2023-11-20:
Thankyou For your response

closed 2023-11-20 by Saiuday9787